openbikesensor/openbikesensor.github.io #8: Interaktiver Schaltplan

boldt, 2020-10-26:


opatut, 2021-03-24:
@boldt Can you describe what you mean? Is there certain plugins or generators that provide this, and how would you keep it up to date with the actual schematics?

Shouldn't linking to the PCB design repository be the simplest and most up to date way of displaying it? Everybody can download the project and KiCad.

opatut, 2021-04-22:
We could use [KiAuto](https://github.com/INTI-CMNB/KiAuto), and a git submodule, to automatically generate up-to-date PDFs from the schematics when building the site, then provide those for download. Might be tricky to integrate into the build workflow, but should be possible. KiAuto seemed to work fine for me in a quick test.