openbikesensor/openbikesensor.github.io #12: Wrong pictures in "Detailierte Aufbauanleitung für den PCB 00.03"

stefferber, 2020-11-29:
The generated "Detailierte Aufbauanleitung für den PCB 00.03" out of  [pcb_00.03.xx_building_long.md](https://github.com/Friends-of-OpenBikeSensor/Friends-of-OpenBikeSensor.github.io/blob/master/hardware/pcb_board/pcb_00.03.xx_building_long.md)
contains pictures from "PCB 00.02" board.

tobst, 2020-11-29:
Maybe we should fix all those files quickly by wrapping almost all of the text in the _include directory until it is really adapted for both versions instead of a warning at the beginning.

stefferber, 2020-11-29:
As a beginner and having access only to PCB 03 this is difficult to do myself.

tobst, 2020-11-30:
Fixed by 882cda2d681f5c394440a76ca1c864062e18d96c

closed 2020-11-30 by tobst