openbikesensor/openbikesensor.github.io #37: Add icon and better favicon

opatut, 2021-03-02:
We should change the header a little to include the icon. Also make the brand text not all uppercase (looks better).

Furthermore, change the favicon to the simplified variant (found somewhere in slack) with just "OBS" instead of "Open Bike Sensor" in it, because it is not readable with the full text. We can keep the bigger icons as-is, they are used e.g. in shortcuts.

Also try placing the high-res (SVG) icon in the title page stage of the homepages (both languages).

ZaidKhan144, 2021-03-02:
Hi, I can work on this issue, but I'm not much familiar with the Hugo framework. Maybe with your little help, I can fix it.

opatut, 2021-03-02:
Write to me in slack (@opatut) if you have specific questions. Otherwise, check out the readme on how to install this locally, and hugo docs for general howto. It's mostly replacing image files, and then adding the image to the `content/de/_index.html` and `content/en/_index.html` files. I have no idea how to add the icon to the header, maybe you'll have to dig through the theme code.

ZaidKhan144, 2021-03-04:
I have installed this locally and ran the app as well.

Nevertheless, I am sorry I wasn't able to find any header file either I have to spend a lot of time figuring out the structure of the themes but couldn't find a related one. 

If anyone wants to work on this, feel free to dive in.

opatut, 2021-09-12:
Related #64.