openbikesensor/openbikesensor.github.io #43: Replace / fix "Ideal Image Slider"

M-a-r-i-a-n-n-e, 2021-03-03:
Using the "alt" attribute for Ideal Image Slider leads to double captions (once a caption then the same text again) on images. The "title" attribute does not seem to work.

This comes from a code that is no longer maintained and was copied from this [repo](https://github.com/Codeinwp/Ideal-Image-Slider-JS/blob/master/extensions/captions/iis-captions.js) to iis-captions.js von assets/js/slider.

`captionContent += '<div class="iis-caption-title">' + slide.getAttribute('title') + '</div>';`

@opatut what should we do about it? Just commenting this line would fix the problem of duplicate titles, but using copy of a library that is unsupported since 2015 might involve other risks, too.

opatut, 2021-03-03:
Well I only used the slider because it was used by the previous jekyll plugin. I didn't mind using it again, I like the way it works in the end. I think we can just hack around in it, if it is unmaintained then at least we won't have to update it :joy: On the other hand, if you want to replace it entirely with something else, that's fine for me as well. I just don't think the effort is worth it right now.

If you hack on this anyway, can you take care that the `alt` attribute on the images are set correctly, even if you switch to a different attribute (title) for setting the caption contents?

M-a-r-i-a-n-n-e, 2021-03-03:
I would prefer to throw it out.. I'm just looking into [carousel](https://mdbootstrap.com/docs/standard/components/carousel/), which already seems to be in the workspace.

opatut, 2021-03-03:
If you want to do that, go ahead! As long as it doesn't involve jQuery ;)

Oh, and if you do, please take care of the print view as well, the images have to print sequentially with captions, because that's used quite often to print out instructions.

M-a-r-i-a-n-n-e, 2021-03-03:
Good point! Maybe we fix this one line first by a workaround (see pull request #44 ) and take our time for including and testing an alternative.