openbikesensor/openbikesensor.github.io #48: Do not include remote resources

opatut, 2021-03-10:
We should get rid of

* google analytics (!)
* google fonts
* bootstramp, jquery, ... from CDNs

Use the network tab to inspect which resources are used.

SubOptimal, 2022-01-30:
Even #128 addressed the load of Open Sans fonts; there are still a few fonts loaded from https://fonts.googleapis.com/. To disable this behaviour, we could wait until https://github.com/google/docsy/pull/869 get merged or apply the following workaround at build time.

```
diff --git a/assets/scss/main.scss b/assets/scss/main.scss
index 1992c27..be42170 100644
--- a/assets/scss/main.scss
+++ b/assets/scss/main.scss
@@ -60,5 +60,4 @@ footer {
     }
 }
 
-@import "rtl/main";
 @import "styles_project";
```

It should not break anything, as we don't use any RTL font on the page.

opatut, 2022-01-31:
It is even simpler, just adding an empty file at `assets/scss/rtl/_main.scss`. I'll see into it.

commit 23d14647 (2022-01-31): Disable Google Analytics tracking.

partly fixes #48
closed 2022-01-31 by opatut