openbikesensor/openbikesensor.github.io #52: Error building site: TOCSS: failed to transform "scss/main.scss"

boldt, 2021-03-14:
Ich bekomme das Projekt bei mir nicht gebaut.

Hugo habe ich bei mir unter **Ubuntu 18.04.** über Snap installiert:

```
$ sudo snap install hugo 
$ hugo version
Hugo Static Site Generator v0.80.0 linux/amd64 BuildDate: 2020-12-31T20:00:21Z
```

Dann bin ich der README gefolgt. Ich hänge nun beim Schritt `hugo server -D`;

```
$ hugo server -D
Start building sites … 
Built in 3274 ms
Error: Error building site: TOCSS: failed to transform "scss/main.scss" (text/x-scss). Check your Hugo installation; you need the extended version to build SCSS/SASS.
```

Mir ist - auch nach einigem Googlen - nicht klar, was das Problem ist. Ich habe das folgende Paket gefunden uns installiert:

* https://www.npmjs.com/package/hugo-extended

Das hat leider auch nicht geklapp.

opatut, 2021-03-15:
* Hast du postcss und co via `npm install` installiert? Liegen die jetzt in node_modules rum?
* Hast du alle submodules rekursiv initialisiert? `git submodule update --init --recursive`

Ich hab auch den hier:

```
Hugo Static Site Generator v0.80.0/extended linux/amd64 BuildDate: unknown
```

opatut, 2021-03-15:
Evtl musst du auch mit snap statt npm das hugo-extended installieren, siehe https://gohugo.io/getting-started/installing/#snap-package:

```
snap install hugo --channel=extended
```

SubOptimal, 2021-07-14:
Die Hugo Installation besteht lediglich aus einem einzelnen Executable (+ README.md und LICENSE). Es ist daher möglich Hugo auch ohne Snap zu verwenden.

- Hugo Archive finden sich unter https://github.com/gohugoio/hugo/releases
- entsprechendes Release auswählen
- unter "Assets" das Archiv entsprechend Betriebssystem und CPU Architektur wählen, die "hugo_**extended**_*" Version herunterladen
- das enthaltene Executable in ein Verzeichnis entpacken, das im Suchpfad (Environment Variable PATH) enthalten ist
- OpenBikeSensor Webseite bauen wie in der [README.md](https://github.com/openbikesensor/openbikesensor.github.io/blob/main/README.md) beschrieben

Beispiel Installation für den aktuellen User unter Linux (das Verzeichnis für User Executable kann evtl. anders lauten)
```
# in temporäres Verzeichnis wechseln
~$ cd /tmp

# Archiv herunterladen
/tmp$ curl --remote-name --location https://github.com/gohugoio/hugo/releases/download/v0.85.0/hugo_extended_0.85.0_Linux-64bit.tar.gz

# Executable in das Verzeichnis für User Executable extrahieren
/tmp$ tar xf hugo_extended_0.85.0_Linux-64bit.tar.gz --directory ~/.local/bin hugo

# Hugo Version prüfen, es muss die Version mit "+extended" ausgegeben werden
/tmp$ ~/.local/bin/hugo
hugo v0.85.0-724D5DB5+extended linux/amd64 BuildDate=2021-07-05T10:46:28Z VendorInfo=gohugoio
```

Wird nicht die _extended_ Version verwendet, dann kommt es zu obiger Fehlermeldung. Mit dem Hinweis die extended Version von Hugo zu verwenden. 

> Error: Error building site: TOCSS: failed to transform "scss/main.scss" (text/x-scss). **Check your Hugo installation; you need the extended version** to build SCSS/SASS.

opatut, 2021-09-12:
@boldt Have you tried it again with extended hugo? Did that work then?