openbikesensor/openbikesensor.github.io #54: Error: Filename too long / build instructions image

madiko, 2021-03-23:
Hey there, 

cloning the repository I got the following error. Would you please shorten the name? Thank you!

> error: unable to create file
[content/en/docs/hardware/v00.03/build-instructions/images/PCB00.03.10_20_Zusammenbau_04_Abschirmung_vorbereiten_8-10Lagen_Alufolien_mit_Klebeband_isoliert_oder_eingeschweisst.jpg](https://github.com/openbikesensor/openbikesensor.github.io/blob/main/content/en/docs/hardware/v00.03/build-instructions/images/PCB00.03.10_20_Zusammenbau_04_Abschirmung_vorbereiten_8-10Lagen_Alufolien_mit_Klebeband_isoliert_oder_eingeschweisst.jpg): Filename too long

Furthermore the name contains german*. Therefore I suggest to rename it with:
PCB00.03.10_20_assembling_04_preperating-shielding

[ * ] I see, that this is true for all the pictures! 
(I thought we did say documentation for developers is in English?)

Thanx a bunch!

First commit coming soon nevertheless.

Best regards,
Franziska

opatut, 2021-03-23:
Well filenames are messy since we never changed them. Sure, we sould update all to English. That would be a different ticket though ;)

I renamed it directly on main branch for you. Can you try cloning again, to see if more files have the issue?

amandel, 2021-03-23:
Might be 'git config core.longpaths true' helps. As I remember this is a Windows thing.

madiko, 2021-03-23:
I couldn't solve the problem remotely. So I merged my fork and started from scratch.
Now everything worked fine (as it should do "out of the box"). Thanks for the fast reply!!

Will open up a new issue. Might help later with solving it. Thanks for giving the ok.
(As soon as the main task is done we can solve this)

closed 2021-03-23 by madiko

opatut, 2021-03-23:
> So I merged my fork and started from scratch. Now everything worked fine

Well I would have expected that, since I fixed it for you:

> I renamed it directly on main branch for you.