openbikesensor/openbikesensor.github.io #61: Embed SWR video #besserRadfahren about OBS

ChrizKu, 2021-03-28:
The [SWR item](https://www.swr.de/radfahren/bedraengt-geschnitten-und-beschimpft-100.html) could be interesting for our startpage or other sites.
They offer also the [embed code for the player](https://www.swr.de/~embed/radfahren/bedraengt-geschnitten-und-beschimpft-100.html). Embed with player could be the better solution in legal perspective, because there are copyright informations in the sourcecode of the player. We could consider to show the [video below](https://www.swr.de/~embed/home/sehr-viele-radfahrer-fuehlen-sich-durch-ueberholende-fahrzeuge-gestresst-100.html) with the interview with _Verkehrsökologe Eckart_ in a smaller iframe, too.

It's [URL](https://pdodswr-a.akamaihd.net/swr/home/1433493.l.mp4) is different to the [Brisant piece](https://odmdr-a.akamaihd.net/mp4dyn2/c/FCMS-c8fe8375-c387-43f1-b9ad-0349fee5c2cf-be7c2950aac6_c8.mp4) and has no "brisant branding" – I assume, that it won't be depublished within 7 days.

ADFC Essen also [embedded](https://essen.adfc.de/neuigkeit/welchen-abstand-halten-kfz-fahrende-beim-ueberholen-von-radfahrenden-ein) direktly the [mp4](http://wdrmedien-a.akamaihd.net/medp/ondemand/weltweit/fsk0/231/2316556/2316556_30771550.mp4) [from ARDmediathek](https://www.ardmediathek.de/video/quarks/mehr-sicherheit-fuer-fahrradfahrer/wdr-fernsehen/Y3JpZDovL3dkci5kZS9CZWl0cmFnLTljNzI2MWE5LTFlNzgtNGRjNC05ZjYwLTMzYWU3YjYzYTMwNw/) (available until 10.12.2025 – so we should ask SWR about their expire date), there is added a adfc URL parameter to the embedded [video url](https://wdrmedien-a.akamaihd.net/medp/ondemand/weltweit/fsk0/231/2316556/2316556_30771550.mp4?enablejsapi=1&autoplay=0&origin=https://essen.adfc.de). It could be found [on YouTube](https://www.youtube.com/watch?v=3NCHngkh64U), but they didn't use it.
I found it in their [twitter reply](https://twitter.com/ADFCEssen/status/1376068501069295618).
Quarks/WDR used laser and a couple of cameras :P

opatut, 2021-05-04:
I think we should write a blog article about this, we can backdate it too and either embed or at least link to the video.

I wouldn't necessarily put it on the frontpage.

opatut, 2021-05-04:
Embed code of the SWR item:

```
<iframe src="https://www.swr.de/~embed/radfahren/bedraengt-geschnitten-und-beschimpft-100.html" width="640" height="360" frameborder="0" webkitallowfullscreen mozallowfullscreen allowfullscreen></iframe>
```

closed 2021-05-09 by opatut