openbikesensor/openbikesensor.github.io #72: Document updating the firmware

opatut, 2021-05-04:
The update mechanism drastically changed between 0.5, 0.6 and 0.7. We should document how to perform a normal firmware update, fresh installation, and also how to do the special upgrades in between the above versions.

opatut, 2021-05-04:
@amandel Maybe you can start, you know which restrictions exist. If you want I can polish it later.

amandel, 2021-05-04:
Quick answer points to https://github.com/openbikesensor/OpenBikeSensorFirmware#updating and https://github.com/openbikesensor/OpenBikeSensorFirmware/blob/master/docs/software/firmware/initial_flash.md#initial-esp-flashing 

I can try to give it a better start but for now this is what I have 8-)

opatut, 2022-03-27:
Initial installation is very well documented now, we should include upgrading from the Web UI.