openbikesensor/openbikesensor.github.io #73: Prominent email button

opatut, 2021-05-04:
Separated out of #64 for discussion.

opatut, 2021-05-04:
@tobst You originally requested this. Please specify:

* Should the email button be a `mailto:` link? If so, expect some more spam on the linked address. If it's a Google account, that spam filter is probably good enough, but nonetheless -- it's a decision that has to be made.
* Where should the button go?
* What email address should be linked?
* Should the button also pre-configure a subject line in the mail client? This is possible with a href like `mailto:foo@example.com?subject=Hello%20World`, in most email clients (e.g. Thunderbird does support it).

boldt, 2021-05-05:
Ich habe sehr gute Erfahrung mit **spamspan** gemacht. Diverse Mail-Adressen, welche damit auf Webseiten veröffentlicht sind, sind de-facto Spam-frei.

Vlt. bekommt man das ja auch in Hugo integriert.