openbikesensor/openbikesensor.github.io #82: Clarify name of the devices vs name of the project

opatut, 2021-06-07:
If we are to *not* use the name "OpenBikeSensor" for the device itself, and only use it to refer to the project and the idea in general, then what do we call the device?

* I've been using "Überholabstandsmesser" ("passing distance measurement device") a bit, but that is ugly.
* Do we even want to distinguish in the dcoumentation, or would it be easier to just call it "the sensor/OpenBikeSensor"?
* In that case, it is also not a sensor, technically. It is a computation device that includes some sensors.

amandel, 2021-06-07:
Feels more easy in english to be able to distinguish:

english | german
-------- | --------------
OBS (Project) | OBS (Projekt)
OBS Device | OBS Gerät(?)
OBS Portal | OBS Portal
OBS Firmware | OBS Firmware
... | ...

opatut, 2021-06-07:
Only... "OpenBikeSensor Gerät" sounds weird in most cases...

amandel, 2021-06-07:
And if we stick to "OpenBikeSensor Device"? Lets try: 

_Wenn du einen OpenBikeSensor Device bauen möchtest, musst du zunächst..._

An anderer Stelle: _wichtiger Bestandteil des Gerätes ist_ -> _wichtiger Bestandteil des OpenBikeSensor Device ist_

We need also other opinions.

nosch2021, 2021-11-30:
also ich mach es ähnlich wie @amandel es beschreibt.
Außer dem Gerät, das habe in meinen Präsentationen immer nur **O**pen**B**ike**S**ensor genannt. Das ist halt der Sensor, das zentrale Element des ganzen Projektes (zumindest aus Radler Sicht). Ich würde den add-on "device" oder "Gerät" nur sehr ungern verwenden (das  klingt sonst wie in einem Verwaltungs-dokument).

opatut, 2022-03-27:
Es hat sich wohl dynamisch nun ergeben dass das Projekt "OpenBikeSensor Projekt" und das Gerät "OpenBikeSensor" heißen.

closed 2022-03-27 by opatut