openbikesensor/openbikesensor.github.io #100: Modellauswahl

Hans-ChristianBenedict, 2021-08-30:
Hallo,

ich hab nicht herausgefunden, wer für den Inhalt auf der Website (Text) verantwortlich ist.
Daher die Frage: Was unterscheidet die einzelnen Modelle des OBS?
Diese Info würde ich dann gerne auf die Website stellen (oder stellen lassen von dem/der, der sie pflegt).

Danke für Feedback und Grüße
Christian Benedict
GarageLab Düsseldorf

SubOptimal, 2021-09-01:
> Was unterscheidet die einzelnen Modelle des OBS?

- Wenn deine Frage zu den Versionen die Unterschiede zwischen
https://www.openbikesensor.org/docs/hardware/v00.03.12/
und
https://www.openbikesensor.org/docs/hardware/v00.03.10/
meint. Das sind unterschiedliche Entwicklungsstände, die sich nach der Versionsnummer des PCB richten. Zudem gab es Änderungen am 3D Gehäuse.
Änderungen fließen ein ...
- aus Rückmeldungen von Interessenten, die einen OBS zusammengebaut haben und dabei auf Schwierigkeiten beim Zusammenbau gestoßen sind
- andere verwendete Bauteile, z.B. THT statt SMD Widerstände

> Diese Info würde ich dann gerne auf die Website stellen (oder stellen lassen von dem/der, der sie pflegt).
Beides ist möglich. Wenn du dich mit dem Webseiten Framwork [Hugo](https://gohugo.io/) und GitHub auskennst, dann kannst du die Änderungen selbst vornehmen und einen Pull Request stellen. Im anderen Fall kannst Du den Text auch in einem Textformat erstellen und es wird sich ein Freiwilliger finden, der es dann in die Webseite einpflegt.

Hans-ChristianBenedict, 2021-09-01:
Hallo Frank, grüße Dich,

danke für Deine Antworten - auch im Slack.

"Änderungen fließen ein ..." ist mir durchaus bewusst. Im ersten Schritt konnte ich das aus der Website nicht erkennen.
Die Unterschiede sind mir jetzt etwas klarer.

"Wenn du dich mit dem Webseiten Framwork Hugo und GitHub auskennst, dann kannst du die Änderungen selbst vornehmen und einen Pull Request stellen. Im anderen Fall kannst Du den Text auch in einem Textformat erstellen und es wird sich ein Freiwilliger finden, der es dann in die Webseite einpflegt."

Ich kenne mich leider nicht mit Hugo aus und nur sehr rudimentär mit GitHub. Über diese Wege wird das also nix.

Text für die Website als Datei: Ich erkenne, dass ich mich deutlich länger in das Projekt einarbeiten müsste um die Unterschiede so klar benennen zu können, dass ich sie als Anleitung veröffentlichen lassen könnte.
Dafür fehlt mir leider die Zeit. Unser Ziel ist es die Sensoren zu bauen und damit Daten zu generieren.

Ich merke aber auch grad, dass ich eigentlich keine Lust darauf habe, erst lernen zu müssen, wie man in diesem Projekt mitspielen darf. Daher nur als Feedback an "Dich" und wen auch immer im Hintergrund des Projektes: Für einen Einsteiger wäre es super, wenn klar erkennbar wäre, wer was wielange hierbei macht und wer welche Skills hat. Von Dir weiß ich jetzt, dass Du 1 Jahr dabei bist - aus einer Reaktion in Slack, nach mehreren Tagen. So einen Info könnte ja auch schon vorab auf der Website stehen, z. B. unter Team? Vielleicht gibt es ja auch Gründe, dass das nicht passiert - aber es wirkt einfach abstoßend, verwirrend, unnahbar. Also auf mich zumindest. Und ich glaube auch, dass es die Einstiegshürde unnötig hoch ansetzt.

Möglicherweise interessiert Dich das auch alles nicht und das hier ist der falsche Kanal dafür.

Dennoch vielen Dank Dir für Dein Feedback und viel Erfolg noch mit den OBS.

Alles Gute
Hans-Christian

closed 2021-09-01 by Hans-ChristianBenedict