openbikesensor/openbikesensor.github.io #102: Website -> Community: Ansprechpartner nicht erkennbar (auch nicht im Slack)

Hans-ChristianBenedict, 2021-08-30:
Hallo in die Runde,

ich versuche seit Tagen im Slack mal herauszufinden, wer denn wer ist in diesem Projekt.
Auf der Website finde ich unter Community (https://www.openbikesensor.org/community/) den Verweis auf Slack. Dort angemeldet und nachgefragt -> (bisher) keine Reaktion. Ich versuche es daher mal über diesen Kanal.

Ich würde mich freuen, wenn klar erkennbar wäre, wer für was wie ansprechbar ist bei diesem Projekt. Aktuell ist mir das auch nach längerem Suchen nicht klar. Ich verstehe auch, wenn es dafür einen Grund gäbe (z. B. keine dauerhaften Kontaktanfragen ;-), aber für Einsteiger - wie mich - macht es das echt schwer, zu verstehen, wie sich dieses Projekt koordiniert/organisiert.

Konkret haben wir im GarageLab eine Anfrage für den Bau von 20 OBS. Wir sind dazu in der Lage und können selbst die Gehäuse drucken und Workshops veranstalten. Wir haben auch einen Kontakt zum ADFC und der lokalen Presse. Dennoch würde ich mich sehr freuen, über kurze Austauschwege im Vorfeld. Einfach für die ein oder andere Frage dazu. 

Danke (wem auch immer) für eine Reaktion auf dieses Ticket.

Viele Grüße
Christian Benedict
GarageLab Düsseldorf

SubOptimal, 2021-09-01:
Da dieses Projekt von Interessierten in ihrer Freizeit entwickelt wird, kann es durchaus sein, dass auf eine Anfrage nicht innerhalb kürzester Zeit eine Antwort/Reaktion erfolgt.

Jeder, der sich für das Projekt interessiert und mitwirken möchte, kann sich einbringen. Mögliche Bereiche gibt es viele, z.B. Dokumentation verbessern, Firmwareanpassungen, Änderungen am Gehäusedesign, einen gefundenen Fehler melden, etc.

Wenn ihr 20 OBS bauen wollt, dann könnt ihr [hier](https://www.openbikesensor.org/bauteile) die aktuellste Teileliste finden. Fragen dazu können im OBS Slack Workspace, Channel #obs_hw_bauteil-börse gestellt werden.

Die Kommunikation, solange sie nicht einen speziellen GitHub Issue betrifft, findet hauptsächlich im Slack Workspace statt.

opatut, 2021-09-12:
Der Grund für diese "Unklarheit der Zuständigkeit" ist, dass eben niemand "zuständig" sein soll. Wir arbeiten 100% auf freiwilliger Basis, und es wäre sehr hinderlich für die Motivation, wenn das Gefühl aufkäme, zu irgendwas verpflichtet zu sein. Auch sind wir so flexibler, wenn mal Freiwillige eine Auszeit nehmen (wie haufenweise geschehen über den Sommer), weil du deine Frage nicht "an die für Thema X zuständige Person" sondern "im Slack-Kanal zu Thema X" stellst, sodass dort andere genauso gut antworten können.

closed 2021-09-12 by opatut