openbikesensor/openbikesensor.github.io #107: Angabe der Bauteile für v00.03.12

notudu, 2021-11-05:
Es wäre hilfreich, wenn in der Stückliste z.B. neben den Widerständen die Bezeichnung (z.B. R7) angegeben wäre.
Generell sollte man Stückliste die Bauteile nach der Bezeichnung auf der Platine sortieren und die Werte in der zweiten Spalte dazuschreiben.
Hilfreich wäre auch ein Farbschlüssel.

gluap, 2021-11-08:
Zumindest der Farbschlüssel (sogar eine komfortable Online-Variante) ist [im Schritt](https://www.openbikesensor.org/docs/hardware/v00.03.12/build-instructions/#schritt-4-widerst%C3%A4nde) beim Einlöten der Widerstände verlinkt.

notudu, 2021-11-09:
Gefunden.

opatut, 2021-11-14:
Das halte ich für wenig hilfreich, da a) die meisten Bauteile keinen Bezeichner haben auf der Platine, und b) ohne die Anleitung eh nicht korrekt gebaut werden kann.