openbikesensor/openbikesensor.github.io #108: Erster Schritt ist falsch bei Bauanleitung für v00.03.12

notudu, 2021-11-05:
Der este Schritt ist das Einlöten des Widerstandes R6, der im Bild auch eingelötet ist. Es ist weder klar, dass dieser zuerst rein soll, noch welchen Wert er hat.

Generell fehlt der Hinweis, dass man immer in der Reihenfolge Passive zu aktiven Komponenten einlötet

gluap, 2021-11-08:
Ich denke dass hier das Foto evtl. das Problem ist - Im Text wird der Widerstand in Schritt 4 eingelötet. https://www.openbikesensor.org/docs/hardware/v00.03.12/build-instructions/#schritt-4-widerst%C3%A4nde

opatut, 2021-11-14:
Da gibt es finde ich kein Richtig und Falsch. Wir haben beides ausprobiert, und so war es finde ich einfacher. Die Bilder sind halt noch "von vorher".

Das Problem ist, dass die passiven Komponenten sehr leicht einzubauen sind, die Buchsenleisten und Lademodule eher schwierig. Da ist es einfacher, wenn die Platine noch ganz flach aufliegen kann. Die Widerstände bekommt man nachträglich immer noch rein.

Auf jeden Fall möchte ich vermeiden, dass erst R6 oder R7 beschrieben werden, dann die Buchsenleisten und Lademodule, un dann wieder "zurück" zu den Widerständen gewechselt wird. Das wäre wohl noch verwirrender.

notudu, 2021-11-23:
Der Hinweis oben auf der Seite ist viel wert.