openbikesensor/openbikesensor.github.io #146: Bauteile für v00.03.12

ZoeIngr, 2022-02-22:
Hi, der Link für die "Buck-Boost Modul for LiPo, 3.3V" funktioniert nicht. Hier wäre m.E. eine alternative: https://de.aliexpress.com/item/32909002977.html?gatewayAdapt=glo2deu&spm=a2g0o.detail.1000060.1.5f83711dXTFWVH&gps-id=pcDetailBottomMoreThisSeller&scm=1007.13339.169870.0&scm_id=1007.13339.169870.0&scm-url=1007.13339.169870.0&pvid=a6ace893-22b2-4ca8-8260-997de9078725&_t=gps-id:pcDetailBottomMoreThisSeller,scm-url:1007.13339.169870.0,pvid:a6ace893-22b2-4ca8-8260-997de9078725,tpp_buckets:668%232846%238109%231935&pdp_ext_f=%257B%2522sku_id%2522%253A%252266192554057%2522%252C%2522sceneId%2522%253A%25223339%2522%257D&pdp_pi=-1%253B0.71%253B-1%253B-1%2540salePrice%253BEUR%253Brecommend-recommend

opatut, 2022-03-03:
Hi @ZoeIngr , danke für den Hinweis. 

Hat jemand genau von diesem Link schon einmal bestellt und kann bestätigen, dass die Module funktionieren? Die Idee an den Links ist, dass wir sie getestet haben und wissen, dass da nicht (nur) schlechte oder falsche Ware kommt, was leider bei aliexpress viel zu oft der Fall ist.

ZoeIngr, 2022-03-15:
Hi Paul,

die haben wir bestellt auf Empfehlung von Roger von Radentscheid Essen. Angekommen sind die, sie sind auch, die die wir bestellt haben. Wir werden diese Woche die OBS bauen- also kann ich mehr dazu in die nächsten Wochen sagen ob sie funktionieren. 

Schöne Grüße 
Zoe