openbikesensor/openbikesensor.github.io #161: WLAN-SSID entspricht nicht der MAC-Adresse

Tutmanichtun, 2022-08-23:
Moin zusammen,
in der "Minimal-Konfiguration" steht als Hinweis:
"SSID ist die MAC-Adresse des Gerätes OpenBikeSensor-xxxxxxxxxxxx."
Das stimmt so nicht, wie ich erfahren durfte, als ich diese MAC in meinen Filter eintrug, der OBS aber dennoch nicht ins WLAN kam. ;)
Zur Erzeugung der SSID wird von der MAC-Adresse 1 abgezogen, die einzelnen MAC-Blöcke Gruppenweise gespiegelt und Nullen am Anfang weggelassen. Zwei Beispiele:
|WiFi-SSID|MAC|
|OpenBikeSensor-378B25BF58|58:BF:25:8B:37:01|
|OpenBikeSensor-E040A71AD6B8|B8:D6:1A:A7:40:E1|
Jede WLAN-Scanning-Software kann allerdings die korrekte MAC-Adresse eines Access-Points anzeigen. Daher frage ich mich: warum der Aufwand?

SubOptimal, 2022-09-07:
Danke fürs reporten. Das ist nicht absichtlich so.

Ich werden ein Issue im Repository der [OpenBikeSensorFirmware](https://github.com/openbikesensor/OpenBikeSensorFirmware) aufmachen.

closed 2022-09-08 by SubOptimal

amandel, 2022-09-10:
Ich würde die Dokumentation anpassen. Die SSID wird von der Id des ESP abgeleitet.

SubOptimal, 2022-09-10:
OK. Das beantwortet auch meine Frage in https://github.com/openbikesensor/OpenBikeSensorFirmware/issues/315#issuecomment-1242728132. (hatte deine Antwort hier zu spät gesehen)