openbikesensor/openbikesensor.github.io #179: Quick Start Guide - broken link

abetusk, 2023-03-25:
The link to the "hardware" section under the quoted text:

> This guide assumes that you have a working OpenBikeSensor. If you don’t, start here!

Gives a 404 error page when following the link "[If you don’t, start here!](https://openbikesensor.org/en/docs/hardware/)".

The "hardware" link on the left hand side works.

It looks like there's an issue with the `www` prefix as the (broken) link in the quoted text is to `https://openbikesensor.org/en/docs/hardware/` whereas the (valid) link to the "hardware" tab in the left next column is to `https://www.openbikesensor.org/en/docs/hardware/`.

Just perusing around, it looks like this issue appears in many different locations in the documentation and other repositories where there are references to `openbikesensor.org` and not `www.openbikesensor.org`.

SubOptimal, 2023-03-26:
This seems to be a configuration issue on the web server side.

A request to URL https://openbikesensor.org/docs/hardware/ is answered with a response status code 301 (Moved Permanently). But the redirect location is broken.

```
HTTP/2 301 Moved Permanently
location: https://www.openbikesensor.orgdocs/hardware/
```
The slash after the host is missed.

@opatut Could you please have a look?

opatut, 2023-03-26:
Thanks, I think it's fixed now.

So we should still fix the fact that there are links to non-www URLs in the page itself. But the redirect has been reconfigured now (after moving the domain) and should work again.

SubOptimal, 2023-03-26:
I'm currently having a look for links which are `< ref ..>` and might need to be changed into `< relref ...>`.

PR #181 (2023-04-11): Generates the webpage for host www.openbikesensor.org.
Solves the issue with references which would point otherwise to host openbikesensor.org.

fixes #179
commit 477920e9 (2023-04-11): Generates the webpage for host www.openbikesensor.org.

Solves the issue with relative links which would point otherwise to
host openbikesensor.org.

fixes #179
closed 2023-04-11 by SubOptimal