openbikesensor/openbikesensor.github.io #192: Build pictures for v00.03.12

MyIgel, 2024-01-15:
While building the sensor, I took some pictures which can be used in the documentation of the current OBS (v00.03.12).
Even as they will be "old" at some point when the next board release is out (which i would be happy to test and maybe take photos too!) I think that they are still a valuable resource for assembling the "older" boards (and maybe in parts also the new ones as a reference).

I might create a PR at some point but feel free to resize / crop / adjust (white balance etc.) them as needed, i'm happy to license the images as LGPL-3.0 or another license when that would fit better :)

The images (still an unsorted list) can be found here: [obs.tgz](https://shrd.me/Bp4yK3)

gluap, 2024-01-15:
@MyIgel Thank you for the pictures. If you still have parts I could send you a board for version 1.0.0 - Some of the current pictures in the 1.0.0 manual are really low quality (because I took them from a video after the fact). Feel free to contact me via matrix (the gluap that joined the CCC-FFM public channel  yesterday). I'll leave this open until pictures can be added to the 0.03.12 manual.

MyIgel, 2024-01-31:
@gluap I added the updated pictures, see the links above. If you have some other intermediate storage i can upload them there too :)

gluap, 2024-02-27:
The pictures are great. I'm starting to add them to the docs.

I've been discussing license with others and would like to use them as "CC-By" with "OpenBikeSensor contributors" as the author - If you prefer to be cited by name or alias or prefer CC-By-SA we could also do that.

MyIgel, 2024-02-27:
That's fine for me too (both variations, whatever fits best so most likely cc by obs contributors) :)